madhephaestus/ESP32Servo #4: Broke build

AustinShalit, 2018-09-24:
https://github.com/madhephaestus/ESP32Servo/commit/78ac6fa82b616b8f9f9c3f00a66ae7c053b3c389 broke my build.

```
In file included from c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\vector:60:0,
from .piolibdeps\Esp32SimplePacketComs_ID5766\src/Esp32SimplePacketComs.h:6,
from src\FinalProject.cpp:2:
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
Archiving .pioenvs\esp32dev\lib95e\libESP32Servo_ID4744.amax(const _Tp& __a, const _Tp& __b, _Compare __comp)

^
Archiving .pioenvs\esp32dev\libd70\libPreferences.a
Compiling .pioenvs\esp32dev\lib532\WiiChuck_ID1465\Accessory.cpp.o
Compiling .pioenvs\esp32dev\lib532\WiiChuck_ID1465\Classic.cpp.o
Compiling .pioenvs\esp32dev\lib532\WiiChuck_ID1465\DJTable .cpp.o
In file included from .piolibdeps\ESP32Servo_ID4744\src/analogWrite.h:5:0,
from .piolibdeps\ESP32Servo_ID4744\src/ESP32Servo.h:67,
from src\FinalProject.cpp:1:
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: error: expected unqualified-id before 'const'
min(const _Tp& __a, const _Tp& __b)
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: error: expected ')' before 'const'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: error: expected ')' before 'const'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: error: expected initializer before 'const'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:219:5: error: expected unqualified-id before 'const'
max(const _Tp& __a, const _Tp& __b)
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:219:5: error: expected ')' before 'const'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:219:5: error: expected ')' before 'const'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:219:5: error: expected initializer before 'const'
In file included from c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\vector:60:0,
from .piolibdeps\Esp32SimplePacketComs_ID5766\src/Esp32SimplePacketComs.h:6,
from src\FinalProject.cpp:2:
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:5: error: 'std::min' declared as an 'inline' variable
min(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:246:7: error: expected primary-expression before 'if'
if (__comp(__b, __a))
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:246:7: error: expected '}' before 'if'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:246:7: error: expected ';' before 'if'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:248:7: error: expected unqualified-id before 'return'
return __a;
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:265:5: error: 'max' declared as an 'inline' variable
max(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:268:7: error: expected primary-expression before 'if'
if (__comp(__a, __b))
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:268:7: error: expected '}' before 'if'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:268:7: error: expected ';' before 'if'
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:270:7: error: expected unqualified-id before 'return'
return __a;
^
c:\users\austinshalit\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:271:5: error: expected declaration before '}' token
}
^
*** [.pioenvs\esp32dev\src\FinalProject.cpp.o] Error 1
```

AustinShalit, 2018-09-24:
Actually, looks like the problem is in Esp32SimplePacketComs?

madhephaestus, 2018-09-24:
check the diff on the tags, its just examples. You may want to look at why you have WiiChuck in your compile? Did FreeRTOS sneak in again?

madhephaestus, 2018-09-24:
All libraries are regression tested against the start-of-term final project code.

closed 2018-09-24 by madhephaestus