madhephaestus/ESP32Servo #17: How to setup the first servo to pwm channel 9

Caemostajo, 2020-08-30:
Not really an issue, but didn´t lnow where to ask. 

Currently using the first 8 pwm channels for other purpose, how to setup the first servo to attach to the pwm channel 9 first and assign the next ones accordingly. 

_ thx

madhephaestus, 2020-08-30:
ESP32PWM allows you to select timers to use, but assumes it has unregulated access to all the channels. I would suggest using this library in place of whatever uses ledc functions.

Caemostajo, 2020-08-30:
I have 4 motor drivers currently with ledc so i can control the frequency of each motor. I honestly want to keep using the ledc functions for this. So there is no way to use this library if there is another library using the timers?

madhephaestus, 2020-08-30:
it plays nice with libraries using other timers, ledc is a different story. It spreads the ledc channels across timers. I do something similar, use one timer for my DC motors at 10khz, and one timer for my servos. Heres an example of me using on 2 PID motor with a servo:

https://github.com/WPIRoboticsEngineering/RBE1001Lib/blob/master/examples/FullSystemTest/FullSystemTest.ino#L26

I would suggest you look at ESP32PWM, it wraps the ledc methods, but arbitrates for the LEDc channels with the timers in a common system.

Caemostajo, 2020-09-03:
I´ve decided to fork your library to make it start on the 9th pwm channel.

Can you please tell me where you are managing this so i can try to change it?

thx

madhephaestus, 2020-09-03:
You really do not seem to understand what channels are and how they are
associated with timers. You need to understand what timers are, that there
are 4 timers each with 4 channels. Timer 0 is channel 0,1,8,9. Timer 1 is
2,3,10,11. Timer 2 is 4,5,12,13 And timer 3 is 6,7,14,15. The timers can be
blocked off, but not the channels. Since you have all 0-9 used at the same
frequency, there are *no more* timers available to use for a servo. You
*need* to change to this library entirely, or change what channels you use
to group them on 3 timers, leaving the last timer available for servos.

Ledc will not check the fact that you reuse timers, and when you start to
use servos, other pwms will break as the base frequency of the timer is
changed by the Servo code.

Caemostajo, 2020-09-03:
I see, thank you very much for clarifying how the timers and channels are linked, For now i will group the channels i´m using so it works with the library.

Thx!

madhephaestus, 2020-09-04:
yeah, if you need 9 channels in the legacy LEDC, then instead of using the
first 9, use:

char ledcIndexes[9]={0,1,8,9,//timer one
2,3,10,11,//timer two
4};//timer three

Which uses up timers 0, 1 and 2. To use this library for Servos, then only
allocate timer 3, that would make it possible to use both.

closed 2020-09-05 by madhephaestus