madhephaestus/ESP32Servo #20: PWM on the ESP32 s2 is currently broken

madhephaestus, 2021-04-13:
This issue:

https://github.com/espressif/arduino-esp32/issues/5050

is causing this library not to work on the s2 hardware.

StuartsProjects, 2021-04-13:
Indeed, I went back to basics and discovered the problem. 

One point of note, the way the ESP32S2 core is distributed seems to have changed;

https://www.esp32.com/viewtopic.php?f=19&t=20364

I wonder how different the 'new' core is the the previous one and if other stuff is affected.

madhephaestus, 2021-04-13:
can you try to set https://github.com/madhephaestus/ESP32Servo/blob/master/src/ESP32Servo.h#L84 to 10 and https://github.com/madhephaestus/ESP32Servo/blob/master/src/ESP32Servo.h#L85 to 1024 and see if the servo PWM starts working for you?

lbernstone, 2021-04-13:
@madhephaestus Do you have an S2 dev board to test fixes?

StuartsProjects, 2021-04-13:
I have an ESP32S2 Saola board hooked up on the bench with a scope on the servo pin. 

Changing to ;

#define DEFAULT_TIMER_WIDTH 10
#define DEFAULT_TIMER_WIDTH_TICKS 1024

Gives, using the Sweep.ino example, a pulse period of 250uS.

madhephaestus, 2021-04-13:
@lbernstone I do not yet. THis issue will stay open until there is a fix, and i can verify the fix.

ilush-a, 2021-05-16:
Any update on this?

madhephaestus, 2021-05-16:
@ilush-a this issue is dependant on https://github.com/espressif/arduino-esp32/issues/5050 Once that issue is resolved the the solution published, this issue will go away. The problem is entirely in the lower layer of the stack, not in this code.

NirZamir1, 2021-12-25:
any news about the problem with the esp32s2 ? 
i have now tried it for the first time and i dont know why it doesnt work

nigeljohnson73, 2022-07-08:
Will this be the same for the S3? I'm using a feather and all goes well until about 2 seconds after attach(pin, min, max) when the Hz on that pin drops to zero, and the servo does not centralise as it does on my LOLIN D1 mini (an ESP8266 using the Adafruit Servo library, not this one, I know :smile: ) Up to that point, I'm seeing 50Hz on the pin from bootup.

It seems odd that the core code is broken for over a year and they release a new board with the same broken stuff on it ☹️

kpochwala, 2023-04-19:
I can confirm that the problem exists in ESP32-S3, and fixing `DEFAULT_TIMER_WIDTH` and `DEFAULT_TIMER_WIDTH_TICKS` to

```
#define DEFAULT_TIMER_WIDTH 10
#define DEFAULT_TIMER_WIDTH_TICKS 1024
```

solves the issue.

closed 2024-06-19 by madhephaestus