madhephaestus/ESP32Servo #21: Different behavior of pin 25 and 27 with analogWrite

willie68, 2021-06-16:
Hello,

first of all I would like to thank you for this great lib.
I am currently working on an implementation of the TPS (in Germany "Tastenprogrammierbare Steuerung, Key Programmable Control"), a very small and limited PLC.
This PLC was originally invented with a small Holtek chip. I've already implemented this in Arduino, ATTiny84, micro: bit, and some other MCUs. Now I am implementing this in ESP32. I'm already done with the main parts because I used the same sources as for the ATMega's and switched to your Lib for analogWrite, Tone and the servo implementation.
The "problem", it's not really a problem, is that GPIOs 25 and 27 (which I used for the DAC outputs) behave differently with analogWrite. 27 uses PWM to generate the output signal, but the 25 uses a real DAC.
As I said, this is not really a problem, but is it possible to switch to PWM at pin 25? Maybe as an additional option?

madhephaestus, 2021-12-19:
the ESP32 has DACS on 25 and 26 so this lib defers to that and exposes it through the analogWrite. I can see how that might be annoying if you wanted PWM there. Ill look into making an option to disable...

madhephaestus, 2021-12-19:
I addd ESP32PWM::DISABLE_DAC as a boolean you can set to disable the DAC feature in analogWrite()

closed 2021-12-19 by madhephaestus