madhephaestus/ESP32Servo #23: Installing ESP32Servo in the ESP32 env, causes espressif8266 env to fail compile

JamesNewton, 2021-09-16:
I have a project with two configurations, one for the ESP32 and one for the NodeMCU. The built in 8266 servo library works for that built, but I need to install this library for the ESP32... but when I do, then the NodeMCU config fails to build. I get:
```

Compiling .pio/build/nodemcuv2/lib874/ESP32Servo/ESP32PWM.cpp.o
In file included from .pio/libdeps/nodemcuv2/ESP32Servo/src/ESP32PWM.cpp:8:0:
.pio/libdeps/nodemcuv2/ESP32Servo/src/ESP32PWM.h:10:28: fatal error: esp32-hal-ledc.h: No such file or directory

************************************************************************
* Looking for esp32-hal-ledc.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp32-hal-ledc.h"
* Web  > https://platformio.org/lib/search?query=header:esp32-hal-ledc.h
*
************************************************************************

 #include "esp32-hal-ledc.h"
                            ^
compilation terminated.
*** [.pio/build/nodemcuv2/lib874/ESP32Servo/ESP32PWM.cpp.o] Error 1
```

JamesNewton, 2021-09-16:
Even uninstalling the library still caused the build to fail. I had to go into the projects (hidden) .pio/libdeps/nodemcuv2 folder and delete ESP32Servo subfolder from it. No idea how that got put in there in the first place.

madhephaestus, 2021-09-16:
this is a bug in how platformIO manages libraries, outside of scope of this repo.

madhephaestus, 2021-09-16:
A possible workaround to the PIO bug would be to use compiler gaurds around the include of this libraries headers. the Nodemcu is a very different processor to the esp32, and they do not have any underlying libraries in common. This library is an extension of LEDC stack, which does not even exist in the esp8266 stack.

closed 2021-09-16 by madhephaestus

JamesNewton, 2021-09-16:
Can you point to or somehow describe the bug in PlatformIO?

madhephaestus, 2021-09-16:
Libraries poison the cache, basically if a project was ever configured to use a library, its compiled in no matter what until you clear the cache. you basically have to clear the cache when switching configurations. PlatformIO is just not designed to robustly clean up after itself when changing architectures. You work around or deleting the .pio directory is the actual, and AFAIK only, solution.