madhephaestus/ESP32Servo #27: Minimum bit width trigerring a "bit width too big (maximum 14)" error with ESP32-C3

loicreboursiere, 2022-09-20:
When runing the examples of the library with a ESP32-C3-DevKitM1 I ran in the following error : 
`[   111][E][esp32-hal-ledc.c:60] ledcSetup(): No more LEDC channels available! (maximum 6) or bit width too big (maximum 14)`
Even using `setTimerWidth` function didn't change anything.

It appeared that the minimum bit width in your library is equal to `DEFAULT_TIMER_WIDTH` which is 16. 
I'm not sure where it comes from. I've seen here and there on the internet when looking at examples of PWM and ESP32 that the `ledcSetup` function that you rely on can accept bit width lower than 16.

I managed to make your library work with the ESP32-C3-DevKitM1 by doing several modifications :

- I've added `#define MINIMUM_TIMER_WIDTH 8` (it could be a smaller value, but I'm not sure if it makes sense) in ESP32Servo.h
- in the first if/else of the `void Servo::setTimerWidth(int value)` function I've changed the valiue `16` by `MINIMUM_TIMER_WIDTH`
- in the `int Servo::attach(int pin, int min, int max)` I've just commented the lines
`this->ticks = DEFAULT_PULSE_WIDTH_TICKS;
this->timer_width = DEFAULT_TIMER_WIDTH;`
I'm not exactly sure how to handle those last two lines of code together with the following one (`this->timer_width_ticks = pow(2,this->timer_width);`) as those lines are already called in the constructor. But the first two would prevent the my modified `setTimerWidth()` to work.

Modifications are here : https://github.com/madhephaestus/ESP32Servo/compare/master...loicreboursiere:ESP32Servo:master

Let me know what you think and if this fix was the correct way to go !

Cheers !

madhephaestus, 2022-09-21:
it would be best to surround these changes with #ifdef terms so that it is only changing the C3 variants. 

Thank you for taking time to make this library better for everyone! Let me know when it is regression tested and open a PR :)

loicreboursiere, 2022-09-28:
Ok first time I'm doing that, so can you elaborate on how you do a regression test ?
I've used #ifdef in the header file, defining `MAXIMUM_TIMER_WIDTH` alongside `DEFAULT_TIMER_WIDTH`.
I've tried this code with my setup (ESP-C3-DEVKitM1) as well as a Adafruit Feather ESP32 (ESP32WROOM).
Everything seems to be ok for me.
The commit is [here](https://github.com/loicreboursiere/ESP32Servo/commit/b9ae21c948576affa87e2b4e452bfe05a1d09693).
When I know more about regression test, I'll open a PR.

Cheers