madhephaestus/ESP32Servo #29: Does not appear to be compatible with ESP32 S3

aaronw2, 2022-10-13:
I see no output on the PWM pin and have tried several. It looks like there are no restrictions on which pins can be used with the S3 and that there are 8 PWM channels instead of 4.

codingcatgirl, 2023-04-01:
This is fixed as of 0.13.0 (#32)

ClemensGruber, 2023-06-16:
We have still problems with the Heltec WiFi Kit 32, Version 3, (see for the board https://heltec.org/project/wifi-kit-32-v3/ ) a ESP32-S3 board. Our HaniMandl code https://github.com/ClemensGruber/hani-mandl/tree/develop is working with ESP32Servo, version 0.12.1 but no more with version  0.13.0, so we recommend currently to donwngrade to 0.12.1 ESP32Servo, see https://community.hiveeyes.org/t/probleme-mit-servo-lib-version-0-13/4866 .

J-R-V, 2023-08-13:
Problems with new Arduino nano ESP32 board when this library is included.

Steps to reproduce error...
1- Select Ardunio nano ESP32 board.
2- Open a new empty sketch (File / New) and compile ... OK, No errors
3- Include the header at the beginning as shown below:
`
#include <ESP32Servo.h>
void setup() {}
void loop() {}
`
Compile again and, in this case, following error appears:
"... ESP32Tone.h:11:41: error: macro "tone" requires 3 arguments, but only 2 given
 void tone(int pin,unsigned int frequency);"

There is no call to this function, so the library may need to be reviewed to resolve this, I think, compatibility issue.

Thanks in advance for your suggenstions to solve this.

ClemensGruber, 2023-11-17:
Still the same with version 1.1.1, a Servo connected to GPIO33 is not working with lib version 1.1.1 but working with 0.12.1 -- are any pins "disabled" in the meantime by the library? It is wokring with other pins.