madhephaestus/ESP32Servo #35: ESP32 S3 Only 4 servo can work at the same time

adama119, 2023-07-20:
I have tested  many times  with version 0.13.0.  

When I use ESP32 S3 (ESP32 S3 DevKitC) ,  if the number of servo is more than 4, the middle servo doesn't work。
I tested 2 ESP32 S3 boards and got the same results .

But When I use another ESP32  board (ESP32 DEVKIT V1),  The following 6 servo are all working fine

`
#define SL1_PIN GPIO_NUM_4  // fine
#define SL2_PIN GPIO_NUM_5   // fine

#define SL3_PIN GPIO_NUM_2   //not work
#define SR1_PIN GPIO_NUM_21  //not work

#define SR2_PIN GPIO_NUM_16  // fine
#define SR3_PIN GPIO_NUM_17  // fine

#define SPEED 15 
#define MA 2 

void CodelabOS::InitServo() {

  CrazyLog::d(kTag, "InitServo");
  s_left_1_.attach(SL1_PIN, 500, 2500);
  s_left_2_.attach(SL2_PIN, 500, 2500);

  s_left_3_.attach(SL3_PIN, 500, 2500);
  s_right_1_.attach(SR1_PIN, 500, 2500);

  s_right_2_.attach(SR2_PIN, 500, 2500);
  s_right_3_.attach(SR3_PIN, 500, 2500);
  delay(2000);
.....
`

Then
I checked the pin signals with my oscilloscope , the blue channel is SL3_PIN in the code above，PWM is not working.
the yellow channel is the SL1_PIN pin in the above code, it works great.

[image]

I don't speak English very well, so please forgive me.

madhephaestus, 2023-07-22:
we are working on updating this library to fully support the s3 core. @keionbis do you have a branch we could review as a PR? 

In the mean time, here is some code we use to generate 16 servo PWM's as a server: https://github.com/OperationSmallKat/LunaMotherboardFirmware/blob/Esp32-s3-migration/src/ServoServer.cpp

flaviobattimo, 2023-10-09:
Hello, same issue here with different PWMs sharing same frequency.

Not sure but it may be related to:
[https://github.com/madhephaestus/ESP32Servo/blob/1.1.0/src/ESP32PWM.cpp#L79](url)
[https://github.com/madhephaestus/ESP32Servo/blob/1.1.0/src/ESP32PWM.cpp#L87](url)

ESP32-S2 has 2 channels per timer maximum, here the limit is 4.

madhephaestus, 2023-10-09:
Hmm, could you add that parameterization to this pr?

borntoleave, 2023-10-27:
I cannot wait for the solution to this problem. 

I found this [ESP32_ISR_Servo library](https://github.com/khoih-prog/ESP32_ISR_Servo) and activated all 16 PWMs. It seems to generate PWM signals on all pins but is unstable. Do you have any insights?

I'm attaching the test code: [ESP32_MultipleServos.ino.zip](https://github.com/madhephaestus/ESP32Servo/files/13189908/ESP32_MultipleServos.ino.zip)

[image]

helicocrasher, 2023-12-29:
I made a similar experience with the ESP32 S2. 
initially only servo1and servo2 worked but not 3 and 4.
In the meantime I have 8 Servos working perfectly :-)
I found out the the original ESP32 has 16 HW (LED) PWM channels ESP32 S2 has only 8  channels. (S3 also 8!?)

At this time I attach 14 PWM channels and then use  the 8 working ones on 
1st attach call return 0 = channel 0 usable on the S2 (Zero ist not error status - it is the channel nr) 
2nd attach call return 1 = channel 1 usable on the S2
3rd attach call return 8 = channel 8 no HW on ESP32 S2 not working
4th attach call return 9 = channel 8 no HW on ESP32 S2 not working
5th attach call return 3 = channel 2 usable on the ESP32 S2 
...
aftter that I use the the PWM channels 0..7 and all is fine.

As an improvement to ESP32Servo library it would be a cool improvement 
A) when the PWM channels 0..7 whould be attached first
B) when it would  return an error ("-1") when failing to assing an existing PWM channel HW

Thanks and cheers

keionbis, 2026-01-18:
It only took 2.5 years but i finally found the time to sit down and make the changes to allow for all 20 servos on the esp32s3 (technically this can be applied to some others but i didnt look into it). The PR is up now