madhephaestus/ESP32Servo #47: Unable to compile example sketch

maxwellsmonson, 2024-03-19:
On any of the example sketches I am receiving an error.

In file included from /Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.h:10,
                 from /Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32Servo.h:68,
                 from /Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32Servo.cpp:53:
/Users/maxwell.monson/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-ledc.h:42:5: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
   42 |     SemaphoreHandle_t lock;        //xSemaphoreCreateBinary
      |     ^~~~~~~~~~~~~~~~~
      |     xSemaphoreHandle
In file included from /Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.h:10,
                 from /Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:8:
/Users/maxwell.monson/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-ledc.h:42:5: error: 'SemaphoreHandle_t' does not name a type; did you mean 'xSemaphoreHandle'?
   42 |     SemaphoreHandle_t lock;        //xSemaphoreCreateBinary
      |     ^~~~~~~~~~~~~~~~~
      |     xSemaphoreHandle
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In destructor 'virtual ESP32PWM::~ESP32PWM()':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:50:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
   50 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In static member function 'static double ESP32PWM::_ledcSetupTimerFreq(uint8_t, double, uint8_t)':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:57:16: error: 'ledcSetup' was not declared in this scope
   57 |         return ledcSetup(chan, freq, bit_num);
      |                ^~~~~~~~~
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In member function 'double ESP32PWM::setup(double, uint8_t)':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:148:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  148 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:149:30: error: 'ledcSetup' was not declared in this scope
  149 |                 double val = ledcSetup(getChannel(), freq, resolution_bits);
      |                              ^~~~~~~~~
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:153:16: error: 'ledcSetup' was not declared in this scope
  153 |         return ledcSetup(getChannel(), freq, resolution_bits);
      |                ^~~~~~~~~
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In member function 'void ESP32PWM::adjustFrequencyLocal(double, double)':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:170:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  170 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:174:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  174 |                 ledcAttachPin(pin, getChannel()); // re-attach the pin after frequency adjust
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In member function 'void ESP32PWM::attachPin(uint8_t)':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:235:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  235 |                 ledcAttachPin(pin, getChannel());
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp: In member function 'void ESP32PWM::detachPin(int)':
/Users/maxwell.monson/Documents/Arduino/libraries/ESP32Servo/src/ESP32PWM.cpp:263:9: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  263 |         ledcDetachPin(pin);
      |         ^~~~~~~~~~~~~
      |         ledcDetach

dannybloe, 2024-03-24:
What did you do to make this work then? I get exactly the same errors.

maxwellsmonson, 2024-03-24:
I believe I made a new ticket after closing this one out but now I can not find it. I never got it resolved still have the error. From my research it some folks say it has to do with the updated servo library however I tried the older versions and got the same error.

DerSep99, 2024-03-30:
I have the same problem. 
Due to my research the problem might in fact be the library itself. I am using a ESP32-C6 and the additional files you need to run this board on the Arduino IDE is a Addin by Espressif. The problem with that is, that they changed the "esp32-hal-ledc.cpp" and "esp32-hal-ledc.h" files in:

C->Users->(your user)->AppData->Local->Arduino15->packages->esp32->hardware->esp32->3.0.0-alpha3-cores-esp32

 They removed the ledcSetup() and the ledcAttachPin() and combined them.

Here you can look that up: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

If we all want to use this library there are two ways:
1. Use a stable version of the espressif esp32 addin (controllers as the ESP32-C6 and ESP32-H2 wont work with those!)
2. Someone who is abled to get the changes in the librarys done needs to do those updates (my coding skills are not that good, but i'll try it my self as well for educational purposes).

Maybe it's just a matter of time until this library gets updated :)

DerSep99, 2024-03-31:
If you check the Pull requests there is a temporary solution. You have to change some function calls and it works fine with the esp32-c6

maxwellsmonson, 2024-03-31:
That is a good find. I will give it a try. I ended up researching the
micropython script for the servo.h library. Got that working well.

Thanks, Maxwell Monson (360) 359-8510

madhephaestus, 2024-04-02:
The solution for this is being discussed here: https://github.com/madhephaestus/ESP32Servo/pull/45

EvanBottango, 2024-06-03:
Arduino ESP32 3.0.0 has been released and is now the default for new installs in the Arduino IDE / users will be ping'd to upgrade their boards in the Arduino IDE to 3.0.0. That leaves this library broken unfortunately by default now for new installs and updates. Totally appreciate the hard work on this library, just thought I'd add that context here on this issue.

madhephaestus, 2024-06-03:
fixed in 3.0.0

closed 2024-06-03 by madhephaestus