madhephaestus/ESP32Servo #53: arduino esp32 board library 3.0.0 incompatibility

kemodaki, 2024-06-01:
switching to 2.0.17 works fine

arduino IDE output:

ESP32PWM.cpp: In destructor 'virtual ESP32PWM::~ESP32PWM()':

ESP32PWM.cpp:53:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
   53 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
ESP32PWM.cpp: In static member function 'static double ESP32PWM::_ledcSetupTimerFreq(uint8_t, double, uint8_t)':

ESP32PWM.cpp:60:16: error: 'ledcSetup' was not declared in this scope
   60 |         return ledcSetup(chan, freq, bit_num);
      |                ^~~~~~~~~
ESP32PWM.cpp: In member function 'double ESP32PWM::setup(double, uint8_t)':

ESP32PWM.cpp:150:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  150 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
ESP32PWM.cpp:151:30: error: 'ledcSetup' was not declared in this scope
  151 |                 double val = ledcSetup(getChannel(), freq, resolution_bits);
      |                              ^~~~~~~~~
ESP32PWM.cpp:155:16: error: 'ledcSetup' was not declared in this scope
  155 |         return ledcSetup(getChannel(), freq, resolution_bits);
      |                ^~~~~~~~~
ESP32PWM.cpp: In member function 'void ESP32PWM::adjustFrequencyLocal(double, double)':

ESP32PWM.cpp:172:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  172 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
ESP32PWM.cpp:176:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  176 |                 ledcAttachPin(pin, getChannel()); // re-attach the pin after frequency adjust
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
ESP32PWM.cpp: In member function 'void ESP32PWM::attachPin(uint8_t)':

ESP32PWM.cpp:237:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  237 |                 ledcAttachPin(pin, getChannel());
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
ESP32PWM.cpp: In member function 'void ESP32PWM::detachPin(int)':

ESP32PWM.cpp:264:9: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  264 |         ledcDetachPin(pin);
      |         ^~~~~~~~~~~~~
      |         ledcDetach

madhephaestus, 2024-06-01:
see discussion in: https://github.com/madhephaestus/ESP32Servo/pull/45

madhephaestus, 2024-06-01:
duplicate of #47

closed 2024-06-01 by madhephaestus