madhephaestus/ESP32Servo #58: Arduino compile error: void value not ignored as it ought to be

abstractBoyd, 2024-06-15:


abstractBoyd, 2024-06-15:
More info about the board driver version:
FQBN: esp32-bluepad32:esp32:esp32
Using board 'esp32' from platform in folder: C:\Users\****\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0
Using core 'esp32' from platform in folder: C:\Users\****\AppData\Local\Arduino15\packages\esp32-bluepad32\hardware\esp32\4.1.0

CrazyLegoMind, 2024-06-15:
hi, same issue here, with this error code compili fromn vscode and platformio: 
`
Compiling .pio\build\robot_maresciallo\lib7f6\ESP32Servo\ESP32PWM.cpp.o
.pio/libdeps/robot_maresciallo/ESP32Servo/src/ESP32PWM.cpp: In member function 'void ESP32PWM::attachPin(uint8_t)':
.pio/libdeps/robot_maresciallo/ESP32Servo/src/ESP32PWM.cpp:312:42: error: void value not ignored as it ought to be
   success=ledcAttachPin(pin, getChannel());
                                          ^
*** [.pio\build\robot_maresciallo\lib7f6\ESP32Servo\ESP32PWM.cpp.o] Error 1
`

apparently the issue line was changed in commit #4684ebf  reverting `success=ledcAttachPin(pin, getChannel());` to `ledcAttachPin(pin, getChannel());` fixed the issue for me as in the new version ledcattachpin no longer returns a bool

you can find that line at ESP32PWM.cpp:312

alx-1, 2024-06-19:
Hi all,  same issue here, thank you for pointing it out :). Also, @abstractBoyd, I'm working with a couple of Unis on a sensor and robotics platform, maybe we should share examples / code / ideas !  The repo is here : https://gitlab.com/darknessdesigner/cajitaabierta at the moment, the examples are in French and Spanish but it would be great to add English examples. It's using ESPparam as a configuration and communication wifi.  As a final note, there's a wiki that explains some of our design choices : https://wiki.eavmuqam.ca/manifeste_pour_une_approche_ecologique_de_la_pratique_des_arts_numeriques#manifesto_for_an_eco-responsible_approach_to_digital_arts_practice ++ Alexandre

closed 2024-06-19 by madhephaestus

madhephaestus, 2024-06-19:
this should be fixed as of 3.0.5