madhephaestus/ESP32Servo #61: Wrong/Miscalibrated servo angle upon ESP32 Power-Up

generic-beat-detector, 2024-07-11:
Hello!

Nice work with this library! 

I intend to use it as part of a robotics project. However, I'm encountering on small issue. Consider the following snippet that sets up the servo for a 90 degree angular position upon start up:

```
#include <ESP32Servo.h>
Servo servo;
void setup() {
  ESP32PWM::allocateTimer(3);
  servo.setPeriodHertz(50);
  servo.attach(18, 500, 2500); 
  // delay(2000);
  servo.write(90);    
}
```

Upon a ESP32 power-up (e.g. upon connecting the USB cable to the port), the angular position of the shaft jumps to about 135 degrees. However, when I press the RESET button or when I invoke `ESP.restart()` from the code, the position is correctly reset to 90 degrees. I though it could be something to do with adding a slight delay upon `Servo` object initialization, but that didn't work either.

What's going on here?

Regards.

madhephaestus, 2024-07-11:
I have seen something similar when the Arduino toolchains switched from 2.xx to 3.xx The problem seems to bee deep in the toolchain, but the work around is easy: 

set the positions at startup twice.

generic-beat-detector, 2024-07-11:
Yes, I think this issue has nothing to do with ESP32Servo. It appears to be an issue with the underlying infrastructure.

For example, upon an ESP32 power-up, and more often that not, the Arduino monitor displays gibberish and commands typed on the monitor do not get to the ESP controller. It is not until I press the RESET button does the program function as anticipated, i.e. the angle gets reset to 90D on startup and commands typed on the monitor (or sent via serial comms) now get to the ESP controller.

Thanks!

closed 2024-07-11 by generic-beat-detector