madhephaestus/ESP32Servo #78: ESP32Servo.h not working in PlatformIO

IamAbhinav03, 2025-06-28:
Hi

I uploaded a simple sweep code to the Seed Studio Xiao ESP32 C3 board using PlatformIO, and the motor is not responding at all. Tried it with Arduino IDE and everything just worked as expected. Not sure if it's a PlatformIO issue or ESP32Servo issue.

Code

```
#include<ESP32Servo.h>

Servo myServo;

void setup() {
  Serial.begin(115200);
  // Initialize the servo on pin 9

  myServo.attach(20);
  
  // Move the servo to 90 degrees
  myServo.write(0);
  
  // Wait for a second
  delay(1000);
}

void loop() {
  for(int angle = 0; angle <= 180; angle += 1) {
    // Move the servo to the current angle
    myServo.write(angle);
    
    // Wait for a short time to allow the servo to reach the position
    delay(500);
  }

  for(int angle = 180; angle >= 0; angle -= 1) {
    // Move the servo to the current angle
    myServo.write(angle);
    
    // Wait for a short time to allow the servo to reach the position
    delay(500);
  }
}
```

platformio.ini

```
[env:seeed_xiao_esp32c3]
platform = espressif32
board = seeed_xiao_esp32c3
framework = arduino
lib_deps = madhephaestus/ESP32Servo@^3.0.6
```

**ESP32 Servo Version:**  3.0.8
**PlatformIO Version:** 3.3.4
**OS**: MacOS Sequoia 15.5 (M1 chip)

freemanzwin, 2025-07-01:
ESP32PWM.cpp

double ESP32PWM::setup(double freq, uint8_t resolution_bits) {
	checkFrequencyForSideEffects(freq);

	resolutionBits = resolution_bits;
	if (attached()) {
#ifdef ESP_ARDUINO_VERSION_MAJOR
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
		ledcDetach(pin);
		double val = ledcAttachChannel(getPin(), freq, resolution_bits, getChannel());
#else
		ledcDetachPin(pin);
		double val = ledcSetup(getChannel(), freq, resolution_bits);
#endif
#else
		ledcDetachPin(pin);
		double val = ledcSetup(getChannel(), freq, resolution_bits);
#endif

		attachPin(pin);
		return val;
	}
#ifdef ESP_ARDUINO_VERSION_MAJOR
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(3, 0, 0)
	return ledcAttachChannel(getPin(), freq, resolution_bits, getChannel());
#else
		**double val = ledcSetup(getChannel(), freq, resolution_bits);
		attachPin(pin);
		return val;
		// return ledcSetup(getChannel(), freq, resolution_bits);**
#endif
#else
	return ledcSetup(getChannel(), freq, resolution_bits);
#endif
}

eidolonFIRE, 2025-08-15:
ESP32-C3 super mini, platformIO.
I also had to downgrade to 3.0.6 because I wasn't getting any output no matter what I tried.

jalukii, 2025-10-06:
I have the same problem in PlatformIO this module is not working. I'm getting an error. 
`[   650][E][esp32-hal-ledc.c:111] ledcAttachChannel(): Pin 18 is already attached to LEDC (channel 0, resolution 10)
[   664][E][ESP32PWM.cpp:333] attachPin(): [ESP32PWM] ERROR PWM channel failed to configure on pin 18!`

When I downgrade to 3.0.6 it says I want to attach pin 255. I using ESP-WROOM-32.